google/cyanobyte #12: Pseudoyaml - Consider adding delay function

Fleker, 2019-03-19:
There may be a need to manually add delays to execution of a pseudoyaml block.

This can be added as another function.

```yaml
exampleWhileLoop:
    input:
        - x: int32
    logic:
        - i:= 0
        - while: i < 10
            - acc := acc + i
            - i := 1
            - delay: 500 // milliseconds by default?
        - return: acc
```

Fleker, 2020-04-24:
Having a delay function would have implications for multithreaded environments and may require the addition of locks or mutexes to achieve.

polfeliu, 2021-03-07:
These environments could be handled with the templates themselves. If I'm using FreeRTOS, i wouldn't use a generic.c, it should be a specific FreeRTOS template.
For IC's that have some internal conversion that you have to wait for it to complete to gather results would be helpful

Fleker, 2021-03-09:
Yes the schema needs a way to define some sort of blocking operation with a callback for once that operation is ready. I don't want to necessarily put it as a direct command, as similar syntax would probably be needed for other kinds of callback operations.

I've recently been considering a different format for defining callbacks:

```yaml
- i := 1
- delay:
    for: 500 # ms
    after:
        - i := 2
        - # Other operations
```

This way, a different type of callback can be specified depending on the IC such as an interrupt or ready flag. Then, different environment like FreeRTOS can better incorporate callbacks into the rest of the system depending on how they want to do scheduling.

The list of standardized statements is something that still needs to be incorporated, and I know that the current complexity is not properly reflected in the schema. This is something I hope to rectify when I get the chance.

polfeliu, 2021-03-09:
Good idea,
Maybe, the after statement could point to another function. Or maybe, this could be better done by adding a `call` statement, to reuse functions.
```
- i := 1
- delay:
    for: 500 # ms
    after: '#/functions/dostuff'
```
Perfect, I look forward to more details to the schema. I'm working on a project that heavily uses the schema, the more info it has the better. I hope to have sound enough to show in a couple of weeks :)

Fleker, 2021-03-11:
I've defined all existing operations in the JSON schema at https://github.com/google/cyanobyte/pull/254

Future operations like delay can be added in a followup.

While calling a function from within a function can be useful, functions do end up getting exposed and not in every case should these be added to the public API.

Instead, a separate operation could be called as one type of `SingleOperation` which can be used elsewhere:

```yaml
- delay:
    for: 500 # ms
    after:
      - i := 2
      - function:
           name: '#/functions/dostuff'
           parameters:
             - i
```

Fleker, 2021-03-29:
The support for delay was merged in #258

closed 2021-03-29 by Fleker