google/cyanobyte #23: Add copyright year to spec info

Fleker, 2019-03-20:
To do the generation of copyright headers, one needs to know both the copyright owner and the year of the copyright. Both can be represented as strings that we plop right into the header, as shown in #20.

chrisfrederickson, 2019-03-20:
Should that be an entirely separate block in the yaml?

```yaml
info:
    copyright:
        year: 2019
        company: Example, Inc.
```

Fleker, 2019-03-21:
Would we want to share the same info as the info.contact field?

chrisfrederickson, 2019-03-21:
Depends on how verbose we want to be. Are there cases where the copyright owner would be different from the maintainer or whoever the point of contact is?

Fleker, 2019-03-21:
It depends. Would we expect an individual to "own" a device spec, while the copyright ownership is the company?

PR #37 (2019-04-29): Add copyright info to spec
Requires a name and date
Generates good license headers

Fixes #23
closed 2019-04-29 by Fleker