google/cyanobyte #25: File extensions for Jinja2 templates

chrisfrederickson, 2019-03-20:
We should use a common file extension for Jinja 2 templates (if only to appease the linters). I'd suggest something like `name.py.jina2` or `name.py.j2`.

Ansible uses `.j2` for what it is worth. https://docs.ansible.com/ansible/latest/user_guide/playbooks_best_practices.html#id11

Fleker, 2019-03-21:
`jinja2` is pretty literal, although `j2` would probably 'fit' better as an extension.

We could also opt to have no file extension, so we'd just have `raspberrypi` and `doc`, which would better fit an aesthetic of a built-in template.

chrisfrederickson, 2019-03-21:
That no file extension is neat, but how would the generator know what extension to use?

Fleker, 2019-03-21:
Right now the Python script just reads in a file as input. If the file isn't a properly formatted Jinja2 template, then Jinja will throw an error. There's not any other validation.

closed 2020-04-03 by Fleker