google/cyanobyte #26: Should registers be contained within the I2C block?

chrisfrederickson, 2019-03-20:
Should registers be contained within the I2C block instead of at the same level? This may make more sense when we support more communication protocols.

Fleker, 2019-03-21:
Would an i2c register be formatted notably different from a SPI register? Is there an instance where a device could support two communication protocols, and create a conflict in knowing how to control registers?

chrisfrederickson, 2019-03-21:
I suppose the other question is, are there chips with multiple serial buses?

Fleker, 2019-03-21:
I've seen occasionally devices that can support one, but allow you to switch, like [this Adafruit I2C/SPI LED Backpack](https://www.adafruit.com/product/292).

I suppose if your target is a microcontroller, you would want to support multiple busses. But then would we want to support more than one i2c bus?

Fleker, 2020-03-18:
In some applications, having more than one i2c bus could be beneficial for redundancy (if one bus gets damaged it can use the backup).

Fleker, 2020-04-16:
Let's keep registers as a separate data model which could be used in different scenarios.

closed 2020-04-16 by Fleker