google/cyanobyte #34: Support for template directories

chrisfrederickson, 2019-03-22:
As the templates increase in complexity, they will almost certainly be comprised of multiple files that will always be used together. We should allow the user to pass in a template directory.

Eventually, do we want to have a set of builtin templates that are referenced by name (e.g. `python` instead of a path to a directory)?

Fleker, 2019-03-22:
Yeah at some point we should have some sort of enum or easy redirect system. In Protobufs, for example, plugins can be specified like:

`protoc --cyanobyte_out=./build foo.proto bar.proto`

This would end up looking for the binary `protoc-gen-cyanobyte`.

While we may need to use multiple template files in the background, I'm not sure we need to pass in a directory as an argument. Jinja allows you to import other files into a base file, so we could have a barebones `templates/raspberrypi.py` file that loads in `templatates/raspberrypi/module1.py` on-demand.

Fleker, 2020-04-02:
Handled by #164.

closed 2020-04-02 by Fleker