google/cyanobyte #40: Codegen for pseudo-yaml

Fleker, 2019-03-24:
Allow codegen to render something like:

A step 1 would allow for simple input/output and logic. Additional support for pseudo-yaml can be part of follow-up.

```yaml
            - asTemp:
                input: test
                variables:
                    - temp: float32
                logic:
                    - if: value > 256
                        - temp:= value
                    - else:
                        - temp:= 0
                return: temp
```

Fleker, 2019-03-24:
To also support future usage of the format, this should be a separate Jinja2 macro that can be inserted into the template.

proppy, 2019-03-28:
If you want to embed code blocks in yaml you might want to take a look at the [block syntax(https://yaml.org/spec/1.2/spec.html#id2795688)

Also curious if you thought of implementing this as a separate layer that reference the register by name allowing the YAML spec to simply describe the register map.

This could also potentially by done in a language different from yaml more suited to pseudo code codegen / data manipulation, ex: https://jsonnet.org/

Fleker, 2019-06-13:
```
  - setVOut:
      title: Set output voltage
      description: set vout
      type: number
      register: '#/registers/EEPROM'
      readWrite: 'W'
      bitStart: 12
      bitEnd: 0
      computed:
        - asVoltage:
            input:
              - vcc: float32
              - output: float32
            logic:
              - output: = output / vcc * 4096
              - send:
                  - '#/registers/EEPROM': output
```

closed 2019-10-07 by Fleker