google/cyanobyte #47: Use of sys.exit in import

chrisfrederickson, 2019-04-01:
In the generated code, it uses sys.exit(1) if the smbus import cannot be found.

https://github.com/google/cyanobyte/blob/bc904d9166cff4397476dd3efd32b848897cf62c/test/sampleData/Mcp4725.py#L26

Is this a bit harsh? If the code is written this way, then it does not allow the end user to deal with the exception themselves.

Fleker, 2019-04-02:
Right now there's a tight coupling to smbus, and without the dependency I'm not sure how the end user will deal with it.

chrisfrederickson, 2019-04-03:
I'm imaging the use case where there is an optional peripheral in a user's program. If the import fails, they disable the functionality in their application.

Fleker, 2019-04-03:
Should I remove the try and just let the exception be thrown in the higher-level code?

chrisfrederickson, 2019-04-03:
Yea

Gadgetoid, 2019-04-10:
It's my experience that `sys.exit(1)` can also break test fixtures and Python auto-doc generation which is why I've been broadly moving away from it in our libraries.

The error/hard exit approach was a hand-holding technique used in libraries written to serve very inexperienced users.

There's no particular reason to tightly couple `smbus` either, since a drop-in-replacement library `smbus2` exists which can be used in its place. It may be preferable to accept a `bus` as a constructor argument, and fall back to `import smbus` where `None` is supplied.

Fleker, 2019-04-10:
At what layer should the library selection be implemented? In the long-term, the peripheral files themselves will be expected to output to Python, C, and other platforms/languages. I wouldn't want to tie that into the peripheral, but you could imagine having two mostly identical templates `raspberrypi.py` and `raspberrypi_smbus2.py` that have different imports.

PR #143 (2020-03-18): Remove use of sys.exit in RPi template
Fixes #47
commit 594b713f (2020-03-18): Remove use of sys.exit in RPi template (#143)

Fixes #47
closed 2020-03-18 by Fleker