google/cyanobyte #57: startBit, endBit, and bits

chrisfrederickson, 2019-04-28:
This issue is for the discussion on startBit, endBit, and bits as it keeps coming up. The key questions to answer are:
1) Should the ordering of startBit and endBit matter? Should we enforce that `startBit > endBit` or the other way around? Should we support both options? https://github.com/google/cyanobyte/pull/38/files#r271088948
2) If `startBit == endBit`, should we add the option to use `bit`? #56

Fleker, 2019-04-29:
Using the phrase "start" and "end" could be conflated in some environments, so it may be a good idea to see if it's possible to handle either direction. Additionally, we don't have a good way to enforce this at the specification level through a schema file.

I do like the simplification of `bit` to indicate a single-bit field.

Fleker, 2019-10-18:
On https://arm-software.github.io/CMSIS_5/SVD/html/elem_registers.html, under the topic */device/peripherals/peripheral/registers/.../fields/field element*, CMSIS has several styles:

* `bitOffset` (LSB) and `bitWidth`
* `lsb` and `msb`
* `bitRange` - A string in the format "[<msb>:<lsb>]"