google/cyanobyte #69: Initialization for Envirophat sensors

Fleker, 2019-08-15:
Sensors have certain init commands which I'm not applying. Ideally these should run at start.

Patil2099, 2019-08-16:
@Fleker Can I work on this? Please, Can you explain it in more detail?

Fleker, 2019-08-16:
Sure, feel free to take the issue.

The issue, in more detail, is this:

To properly initialize certain hardware sensors, like the envirophat color sensor, there are a few bytes that need to be written. We can see this in the manually curated driver: https://github.com/pimoroni/enviro-phat/blob/master/library/envirophat/tcs3472.py

```python
self.i2c_bus.write_byte_data(ADDR, REG_ENABLE, REG_ENABLE_RGBC | REG_ENABLE_POWER)
self.set_integration_time_ms(511.2)
```

In the code, this is defined as a `setup` method.

This always needs to run, and we need some way of defining the registers and values to write. While the current version of [a proposed spec file](https://github.com/google/cyanobyte/pull/68/files#diff-e2908924856e33e100891f36b7b0e147) does define these values:

```yaml
registers:
  - enable:
      address: 0x80
      length: 8
      title: Enable
      description: Enable specific components of the peripheral
functions:
  - init:
      title: Setup the device configuration
      description: Enable RGBC and Power
      register: '#/registers/enable'
      readWrite: 'R/W'
      bitStart: 7 # The whole register
      bitEnd: 0
      type: 'enum'
      enum:
        - Power:
            title: Power
            value: 1
        - RGBC:
            title: Color
            value: 0b10 
```

The output file has an explicit `init` method. Ideally, multiple things should be added to the specification and underlying codegen:

* There is some magic function name, like init, that will get run in the driver constructor
* The function should contain a `computed` element, like in the [BMP280](https://github.com/google/cyanobyte/pull/70/files#diff-e3dd3e33e22cb242461644d3a63ea66f) example, that shows which values to write to which registers
    * This will require changes to both the codegen tool and the spec

```yaml
functions:
  - temperature:
      title: Temperature as read by sensor
      description: Reads the temperature
      readWrite: 'n'
      computed:
        - asRaw:
            variables:
              - valueMsb: int8
              - valueLsb: int8
              - valueXlsb: int8
              - output: int16
            logic:
              - valueMsb:
                - read: '#/registers/TempMsb'
              - valueLsb:
                - read: '#/registers/TempLsb'
              - valueXlsb:
                - read: '#/registers/TempXlsb'
              - output:
                - sum:
                  - bitShiftLeft:
                    var: valueMsb
                    bits: 12
                  - bitShiftLeft:
                    var: valueLsb
                    bits: 4
                  - bitShiftRight:
                    var: valueXlsb
                    bits: 4
            return: output
```

Hopefully that helps.

Fleker, 2020-03-18:
Blocked on #122

PR #149 (2020-03-28): Add special function that runs on construction
With a function `init` and computed of `onCreate`, the constructor
executes the function.
Example done for TCS3472
Not applied to generic C as init function doesn't have fnc ptrs

Fixes #69
commit 2c020b5e (2020-03-28): Add special function that runs on construction

With a function `init` and computed of `onCreate`, the constructor
executes the function.
Example done for TCS3472
Not applied to generic C as init function doesn't have fnc ptrs

Fixes #69
closed 2020-03-28 by Fleker