google/cyanobyte #104: Arduino integration testing

Fleker, 2019-11-11:
Have a way to programmatically build and lint the project for Arduino / C++.

urish, 2020-04-27:
Hi Nick, author of [AVR8js](https://github.com/wokwi/avr8js) here. First of all - thank you so much for creating this project! I've wanted to create something similar for a long time, and I truly wish this project will be successful.

Second, I'd love to help with the Arduino integration testing, if I can. Do you have anything specific in mind yet?

Fleker, 2020-04-27:
Thanks for your feedback.

I'm not entirely sure what a good integration test for Arduino would look like. As part of the CI process, using [Cloud Build](https://github.com/google/cyanobyte/blob/master/cloudbuild.yaml), it has access to [Arduino sample data](https://github.com/google/cyanobyte/tree/master/test/sampleData/arduino). I want to ensure the quality does not change through various iterations of the Arduino template. For the Raspberry Pi I run a linter. For Arduino having a linter or some way to build the code will help me catch potential issues early.

urish, 2020-04-28:
Sure, I will have a stab at that. I'm not familiar with Cloud Build, but it seems to be based on docker, so I will try to set up a container which builds the code for Arduino and gives you the desired feedback.

Can you please assign me to this issue?

PR #201 (2020-04-29): Add automated arduino builds on CI
This should compile all the Arduino code on every commit, and fail if there's any syntax error. 

I also had to add an `arduino.ino` file into the `sampleData` directory, otherwise the Arduino CLI won't build the project.

Fixes #104
closed 2020-04-29 by Fleker