google/cyanobyte #114: Kubos logic has `return`, but the return type is void

Fleker, 2019-12-14:
We should update whatever the value of `val` is to the thing we are supposed to return, rather than return directly.

mandrin17299, 2019-12-14:
I would like to take this up. Could you please help me get started? Thanks!

Fleker, 2019-12-15:
On [line 179](https://github.com/google/cyanobyte/blob/master/templates/kubos.c#L179) of `kubos.c`, the auto-generated function has `void` as the return type.

Later on, the return value appears to be explicitly returned on [line 199](https://github.com/google/cyanobyte/blob/master/templates/kubos.c#L199) and [line 203](https://github.com/google/cyanobyte/blob/master/templates/kubos.c#L203). This return statement is poorly placed, as it shouldn't be a direct return. Rather, we should reassign the value of our pointer `val`, which is defined in the function header.

mandrin17299, 2019-12-15:
Hey @Fleker , is this the change that was required?
[image]

Fleker, 2020-03-04:
Yes @PranshuPanwar that would be the correct fix to the issue. You may need to rebase on top of a few larger changes.

PR #174 (2020-04-07): Fix return logic in Kubos functions
Fixes #114
commit bba1b843 (2020-04-07): Fix return logic in Kubos functions (#174)

Fixes #114
closed 2020-04-07 by Fleker