google/cyanobyte #203: Create a logo

Fleker, 2020-04-30:
The project was recently mentioned in an [Adafruit newsletter](https://blog.adafruit.com/2020/04/29/icymi-python-on-microcontrollers-micropython-turns-7-circuitpython-5-3-0-rc0-is-out-and-more-python-adafruit-circuitpython-icymi-circuitpython-micropython-thepsf/) with a diagram from [the website](https://cyanobyte.dev). The header image was the diagram rather than a more useful logo.

urish, 2020-04-30:
Congratulations for the mention!

Do you need some help with the logo, Nick?

Fleker, 2020-04-30:
I've done minor graphic design in the past, but I'm open to ideas here.

urish, 2020-04-30:
Do you have some general idea of what the logo should symbolize?

Fleker, 2020-04-30:
No I'm not really sure that it should be. The name is a combination of "cyanoprint" and a digital "byte", but how would that translate into a logo?

urish, 2020-04-30:
Google images mostly brings leaves for cyanoprints:

[image]

So perhaps some kind of leaf in 8-bit pixelated style?

Fleker, 2020-04-30:
Like some sort of digital leaf?

urish, 2020-04-30:
Yeah!

Fleker, 2020-05-03:
[image]

Here's my first attempt. I'm still not sure about the design of the peripherals.

urish, 2020-05-03:
Yes, the small details are lost when the logo is turned into a thumbnail:

[image]

If you decide to keep the peripherals, I'd probably add one or two more, and break the symmetry (maybe also use a different shape for each, rectangle, triangle, circle, etc.). In the current configuration they resemble hands

Fleker, 2020-05-07:
Swapped it up a little more to emphasize the leaf with minimal wiring
[image]
[image]

urish, 2020-05-08:
Can you also try without the rectangle? I think it'll improve the contrast of the logo against its background

Fleker, 2020-05-09:
Removing the background

[image]
[image]

Fleker, 2020-05-09:
Change in cable direction

[image]
[image]

PR #264 (2021-03-29): Add finished logo and include in README and devsite
Fixes #203
commit f2560af1 (2021-03-29): Add finished logo and include in README and devsite (#264)

Fixes #203
closed 2021-03-29 by Fleker