google/cyanobyte #243: Pip package imports are invalid

silvergasp, 2020-09-25:
It appears that the import paths used in your pip package do not work as expected.
*Steps to reproduce*
1. Install cyanobyte:
```shell
pip3 install -U cyanobyte
```
2. Run code generation
```shell
cyanobyte-codegen -t ./templates/doc.md -o ./build ./peripherals/MCP4725.yaml 
Traceback (most recent call last):
  File "/home/nathaniel/.local/bin/cyanobyte-codegen", line 5, in <module>
    from cyanobyte.codegen import gen
  File "/home/nathaniel/projects/cyanobyte/cyanobyte/codegen.py", line 33, in <module>
    from convert_json_to_yaml import convert_json_to_yaml
ModuleNotFoundError: No module named 'convert_json_to_yaml'
```

Running directly from the cyanobyte subdirectory in this repository does work, however this seems to defeat the purpose of the pip package.
No output to stdout when run from `{project_root}/cyanobyte` and the generated outputs seem to work as expected
```shell
cyanobyte-codegen -t ../templates/doc.md -o ../build ../peripherals/MCP4725.yaml
```

Fleker, 2020-09-28:
Thanks for reporting this. It's been some time since the pip package was updated with more recent changes, so it's probably worthwhile to make a new release and verify it works as expected.

Fleker, 2021-02-12:
Resolved in version 0.0.2, which should now bundle more content and access it.

closed 2021-02-12 by Fleker