google/cyanobyte #244: BMP280 Generated files for Arduino with the given datasheet is not working?

rohitsam, 2020-10-03:
Hey, 

I have run cyanobyte-Codegen for Sensor BMP280 and generated code for Arduino.

> cyanobyte-codegen -t templates/arduino.cpp -o ./build peripherals/BMP280.yaml

> cyanobyte-codegen -t templates/arduino.h -o ./build peripherals/BMP280.yaml

Later I have used these generated library files and written these lines of code for Atmega 328(Arduino) 
```
#include "BMP280.h"
TwoWire Wire1;
BMP280 sensor(Wire1);

void setup() {
sensor.begin();
Serial.begin(9600);
}
void loop() {
Serial.println(sensor.temperatureasCelsius());
delay(2000);
}
```

But I get null output.
I have also confirmed my  my hardware to be working fine, since I have verifyed it with [adfruits library ](https://github.com/adafruit/Adafruit_BMP280_Library)

rohitsam, 2020-10-05:
In the cyanobyte datasheet Many registers are given wrong datasize #246

rohitsam, 2020-10-07:
```
uint16_t BMP280::readDigP8() {
    uint8_t datum;
    uint16_t value;
    _wire->beginTransmission(DEVICE_ADDRESS);
    _wire->write(REGISTER_DIGP8);
    if (_wire->endTransmission(false) != 0) {
        return -1;
    }

    if (_wire->requestFrom(DEVICE_ADDRESS, 2) != 2) {
        return 0;
    }

    datum = _wire->read();
    value = value << 8 | datum;
    datum = _wire->read();
    value = value << 8 | datum;

    return value;
}
```
This is the generated code of BMP280 for arduino, and  the **endianness** is ignored.
[image]

Fleker, 2020-10-14:
Yeah I'll need to re-review the datasheet and ensure that the registers are all correctly typed.

rohitsam, 2020-10-14:
The template for Arduino doesn't have method to typecast variables.

Fleker, 2021-02-12:
Arduino variables should now be properly signed as-necessary.

closed 2021-02-12 by Fleker