google/cyanobyte #259: Field and register name unique keys and validator

polfeliu, 2021-03-14:
@Fleker I'm wondering if the names of fields or registers are supposed to be unique keys. 
As far as I understand, duplicate keys in YAML is uncommon, and if we are referencing them as "#/registers/RegisterA" it should only point to one instance. 
Many devices have fields in different registers that are equally named. We could agree to simply put the "datasheet name" to the title of the field and the field name be whatever unique name we please.

If we agree that fields and registers name are unique keys the issue is very simple: Validator doesn't throw errors for duplicate keys

Fleker, 2021-03-15:
Registers and fields should be given unique names as in the top-level definition the name of that register should be a unique key.

```yaml
registers:
    RegisterA:
        # ...
```

When referring to it in logic, the prefix is used in part to distinguish it from a variable name or other type of value. But there should still only be a single `RegisterA` or it should not be correctly validated.

Fleker, 2021-03-16:
Ah when I open this in my code editor the editor is highlighting the error. I would expect the validator to catch this, but I will investigate why it's not.

Fleker, 2021-03-27:
The yaml loader will override duplicate keys silently rather than throw.

Further discussion in
https://github.com/yaml/pyyaml/issues/165

commit 64855b59 (2021-03-29): Add duplicate key check in Cyanobyte validator tool (#261)

Has to be an explicit validation as the library silently merges
Polish console output from validator
Fixes register schema

Fixes #259
closed 2021-03-29 by Fleker